kellerza/hassio-sma-em #1: Friendly names not recognised

johnnysmithy852456, 2021-03-12:
Hi Kellerza
The Add-on works great thank you.
I just couldn't see the filtered values after adding the info to the configuration.yaml file
I used the friendlly (sensor.sma_grid_total_5s) names like you show in the docs.md but just shows unavailable. 
So I had to use the original entity name (sensor.pconsumecounter) in the filter platform part.
Not sure why but I thought you would like the feed back.
Thanks Again
Nicko1

kellerza, 2021-03-12:
Great news, thanks for the feedback. And glad you got your connectivity fixed!

in docs.md there is a table with old names& new names. So after discovery by the addon I renamed my sensor’s entity_id to the new name, not just the friendly name. Iirc you click on the gear to change it, maybe that was not clear enough, but the purpose of the table 😄

closed 2021-03-21 by kellerza

kellerza, 2021-06-15:
@johnnysmithy852456 with the new (dev) version of the plugin the filter can be applied at the source and no need for adding them in home assistant. Once you have the addon repo, simply install the dev version and try it out - info on the docs tab of the new version