kellerza/hassio-sma-em #2: Is there a key for "house" consumption?

cpohli, 2021-03-27:
Hi,

first of all: A big **thank you** for getting your handy dirty with SMA!

I recently got a solar system of a SMA TRIPOWER (which i can acces via the HA SMA integration) and a SMA Sunny Island (for the installed battery, which i cannot acces, because there is only one SMA integration possible, right).

I read your comments of some so not reliable date and tried your SMA HomeManager via MQQT here.
The "only" readouts i can get are almost the same ones i get via the SMA integration.

I was hoping for a way to access the battery (via the HomeManager EM), but more than this for a way to monitor the consumtion of my whole house as i can see it in the SunnyPortal webpage. I thought, in the end ALL data is running through the HomeManager and could be read there.

Am i wrong with that assumption? None (TRIPOWER and Sunny Island) are givinhg me a saensor for that, so i figured, it should be possible via sma-em?

Thank you for any information.

kellerza, 2021-03-27:
The SMA [energy meter](https://www.sma.de/en/products/monitoring-control/sma-energy-meter.html) (is the only SMA part being read by this integration SAM-EM) should be able to give you consumption from he grid only (it is wired between the grid and your PV plant&house). It does not have any access to your batteries. Typically the SMA inverter will also read&report the values from the SMA energy meter, so there will definitely be overlap.

cpohli, 2021-03-29:
Hi,

thanks for getting back so quick.
Perhaps i did not choose the right wording.
What i meant is:
The setup is (as you otulined) Grid <-> SMA EM <-> HOUSE <- Solar Inverter <- PV Panels, right.
If i login into the Sunny Portal or the SMA Energy-App i can see an "Actual Energy used in my house"-Reading.
That's the central point there and it is fed by either grid_consumption or pv_yield or battery or any combination of these three.

I wonder, where this "house consumption" is coming fron, as there is no other meter involved (SMA wise) and i cannot spot this on either solar inverter nor battery inverter. And there are only these 3 things involved (2 inverters and the EM).

cpohli, 2021-03-29:
Perhaps this picture (although in german ;-)) ist helping.
I'm particulary interesting in the origin of the central "house"-part here...

[image]

kellerza, 2021-06-17:
@cpohli the SMA energy meter is typically installed between the utility grid and your house. In that case it will measure power imported/exported to the grid. 

This hassio addon only receives readings form the Energy meter, so if you look at `pconsume` and `psupply` you should see the 2.33kW being exported (I'm guessing in psupply)

The other values you should be able to get from the pysma library/ SMA integration in HomeAssistant (i.e. 5.81kW being generated from a Sunny Boy and the 2.84kW being used to charge the batteries form the SMA Sunny Island. The house usage should then be the difference...)

But since the SMA components are modular, you'll have to see how your system is wired and from which components you can get values

Hope this helps

closed 2021-08-05 by kellerza