kellerza/hassio-sma-em #3: Extension stops updating after a few minutes

pabera, 2021-04-06:
I am using this extension with a freshly installed HA. I know the setup is correct because I do get (correct) data from my SMA Tripower. Unfortunately, the data flow only runs for around 4-5 minutes and it stops. The HA entities do not get updated anymore until I restart it with the same result .. it runs for another period of 4-5 minutes. 

Am I missing a setting or something? The logs do not say anything strange. Any help appreciated

kellerza, 2021-04-07:
Hi, there is not much I can do with this level of info, but you could try:
- turn on the watchdog (if the energy meter extension crashes)
- set the debug flag in config (then the log should print readings)
- use mqtt explorer to see if something gets pushed to mqtt

pabera, 2021-04-07:
I have turned on Watchdog. It suppose that the extension actually doesn't crash. 
As soon as I restart the extension, data gets populated, both for the extension itself and MQTT.. I see the values in both logs properly.

> (then the log should print readings)

No errors printed, it just stops populating.

Here are a few lines of the log of MQTT. They won't help much but just wanted to show how it is.

```
1617791441: Client SMA-EM@6a2c1c7a-sma-em disconnected.
1617791447: New connection from 192.168.1.201 on port 1883.
1617791447: New client connected from 192.168.1.201 as SMA-EM@6a2c1c7a-sma-em (p2, c1, k60, u'mqtt').
1617791447: Client SMA-EM@6a2c1c7a-sma-em disconnected.
1617791452: New connection from 192.168.1.201 on port 1883.
1617791452: New client connected from 192.168.1.201 as SMA-EM@6a2c1c7a-sma-em (p2, c1, k60, u'mqtt').
1617791452: Client SMA-EM@6a2c1c7a-sma-em disconnected.
1617791457: New connection from 192.168.1.201 on port 1883.
1617791457: New client connected from 192.168.1.201 as SMA-EM@6a2c1c7a-sma-em (p2, c1, k60, u'mqtt').
1617791457: Client SMA-EM@6a2c1c7a-sma-em disconnected.
1617792818: Saving in-memory database to /data/mosquitto.db.
--- END OF LOG
```

And here the last logs of the extension

```
mqtt: sma-em topic SMA-EM/status/3009876046 data published 10:29:45:{"pconsume": 0.0, "pconsumecounter": 652.2551, "u1": 227.269, "timestamp": 1617791385.2000964}
mqtt: publishing SMA-EM/status/3009876046/pconsume:0.0
mqtt: publishing SMA-EM/status/3009876046/pconsumecounter:652.2551
mqtt: publishing SMA-EM/status/3009876046/u1:227.269
mqtt: publishing SMA-EM/status/3009876046/timestamp:1617791385.2000964
mqtt: ssl disabled
mqtt: sma-em topic SMA-EM/status/3009876046 data published 10:29:50:{"pconsume": 0.0, "pconsumecounter": 652.2551, "u1": 227.371, "timestamp": 1617791390.4012456}
mqtt: publishing SMA-EM/status/3009876046/pconsume:0.0
mqtt: publishing SMA-EM/status/3009876046/pconsumecounter:652.2551
mqtt: publishing SMA-EM/status/3009876046/u1:227.371
mqtt: publishing SMA-EM/status/3009876046/timestamp:1617791390.4012456
mqtt: ssl disabled
--- END OF LOG
```

You can see that time kinda matches up.. and there is no error tell why it's not populating anymore. Are there other places where I could debug errors?

The expected behaviour of this extension is to run endlessly, polling data constantly, correct?

pabera, 2021-04-07:
I believe I need to get into coding myself. I had a look at your `run.py` and it actually looks ok. I need to check if the Docker maybe crashes or if there is any issue with the code itself locally.

If you have any insights from your side, still let me know ;-)

pabera, 2021-04-17:
I wrote my own client in Node to listen to the SMA-EM multicast in my local environment and figured out that the connection drops after several minutes (around 3-6) as well, just like I described above. I am unsure whether this is related to my local network or the SMA-EM in general. I am closing this since I am sure this problem is not related to your code.

closed 2021-04-17 by pabera

kellerza, 2021-06-15:
@pabera you could maybe try the dev version. It is on the same repo

It now uses an asynchronous way to read the data from the socket and it might give better results… it should also allow the addition of a timeout for the read (in which case we can restart the multicast group subscription)

pabera, 2021-06-21:
Thanks for this updated work. I like the fact that you have introduced a threshold.
Unfortunately, my original issue persists. The software reports for 4-5 minutes and then stops. I will check you code and check if I can make PR to have an optional reconnect similar to I have solved the problem in my Node test.