kellerza/hassio-sma-em #5: "unknown" value for sensor.power_l1

MisfitPunisher, 2021-07-16:
Hi firends! When there's not solar production the sensor.power_l1 does not appear with a 0 value, appears as a "unknown". 
And also, all the other sensors appears as a "currently unavailible" when I reboot the HA and there is not solar production.

Can it be solved? Thank you for your integration, I love your work, makes our lives easier!

macnug, 2021-07-29:
Hi i tried everything and i have your same issue. All sensors are in "unknown"

kellerza, 2021-08-05:
Do you see anything in your HA Core logs? Have you tried add `DEBUG: 1` to your add-on configuration? 

The Home Assistant integration https://www.home-assistant.io/integrations/sma/ - using the [pysma library](https://github.com/kellerza/pysma)  shows "unknown" during the evening

The only reason I can think of that you see "unknown" with the Energy Meter HASS.IO Add-On is if it stopped working completely/never worked

macnug, 2021-08-05:
> Do you see anything in your HA Core logs? Have you tried add `DEBUG: 1` to your add-on configuration?
>
> The Home Assistant integration https://www.home-assistant.io/integrations/sma/ - using the [...]

Hi Keller, 
i tried with the DEBUG:1 and i have this error:

DEBUG:asyncio:Datagram endpoint remote_addr=None created: (<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>, <__main__.MulticastServerProtocol object at 0x7f34b4853430>)

Do you have some suggestion on how to fix it?
Thanks

kellerza, 2021-08-05:
Hi @macnug ,
- has it ever worked?
- what environment are you using?

Your error seems related to the multicast socket. It is needed to receive the Energy meter's Speedwire packets,

macnug, 2021-08-05:
> ed to receive the Energy meter's Speedwire packets,

Hi. thank you for your reply.
It worked for months but it stopped working, so i started using only sma solar integration and i deactivate the addon. With these last releases the prysma library removed some sensor that were fundamental like total house consumption. Now there are multiple metering sensor that are "unknown" also during day time. 
I tried with SMA energy meter addon but it doen't work for me. 

my config:
-sma sunnyboy 2,5
-sma energy meter in Speedwire connection
-nuc with proxmox and multilple VM, in one there is Hassio os.

kellerza, 2021-08-05:
You do not need the source IP /IP of the SMA-Energy Meter

What is the old version you used? 0.5.4? I created a "Legacy" plugin based off that branch if you want to try that. https://github.com/kellerza/hassio-sma-em/tree/main/sma-em-legacy

Just add the **SMA Energy Meter (Legacy)** plugin under the same repository in the Supervisor

macnug, 2021-08-11:
Hi,
I tried also the legacy addon. With debug 1 I have this error:
 ifelse: fatal: unable to exec ./run.py: Permission denied

[cmd] ./run.py exited 126

Can you help me troubleshoot “permission denied”??

I tried all the combination available for mqtt and serial…

Thank you so much

Marco

kellerza, 2021-08-12:
The joys of file permissions... you can try 0.5.5

macnug, 2021-08-12:
Hi Keller,
I tried with the last version and I have this error:

DEBUG:asyncio:Datagram endpoint remote_addr=None created:
(<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>,
<__main__.MulticastServerProtocol object at 0x7fd261dd3430>)

Actually I dunno how to fix multicast error.

Have you ever find this time of issue?

Thank you for everything..

Marco

northalpha, 2021-08-12:
Can you confirm that you are seeing the multicast packets on the HA instance or inside the container? Are you aware of how to check tis via tcpdump?

macnug, 2021-08-12:
Never did that, and no I dunno how to check. May you suggest a guide or
some tips on how to do that?
Thank you!!!!

kellerza, 2021-08-16:
@macnug the error still shows asyncio, are you sure you strarted the legacy addon?

This is not an error you will get for the Legacy addon

```
DEBUG:asyncio:Datagram endpoint remote_addr=None created:
(<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>,
<__main__.MulticastServerProtocol object at 0x7fd261dd3430>)
```

closed 2024-07-08 by kellerza