kellerza/hassio-sma-em #6: Compatibility with 2021.8

northalpha, 2021-08-04:
Would it be possible to modify the addon to work with the statistics changes made in 2021.8 that we can use the Energy Dashboard? That would be awesome, so we would be able to get rid off the awful sunnyportal stuff.

kellerza, 2021-08-04:
Have not checked if this is supported by MQTT autodiscovery yet, if that is the case it should be possible

johnnysmithy852456, 2021-08-05:
Hi
Looks like it could just be the device Class power (pcomsume)  HA is only adding entities with a Energy class at the moment.

kellerza, 2021-08-05:
From addon version 2021.8.1 (released 5 min ago) you can change the MQTT autodiscover config that gets sent to homeassistant:

```
AUTODISCOVER_CONFIG:
  - key: state_class
    value: measurement
  - key: last_reset
    value: '2021-07-30T00:00:00+00:00'
```

state_class is the new default when you update

It seems that last_reset is NOT supported by MQTT auto-discovery today, so you can try to use customize to override the entities with that,- refer to [this](https://community.home-assistant.io/t/long-term-statistics-from-mqtt/326990) solution

closed 2021-08-05 by kellerza