kellerza/hassio-sma-em #7: Update in core-2021.6.6 to 0.7 no mqtt

Jogiarea, 2021-08-05:
Hi

I ve updatet sma em in core-2021.6.6 to version 0.7,
The green light shows that 0.7 is running.

But the MQTT entity dosnt work

My config:

MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: XXXXXXX
MQTT_PASSWORD: XXXXXXX
SMA_SERIALS: []
FIELDS:
  - pconsume
  - pconsumecounter:max
  - u1
  - psupply
  - sconsume
  - ssupply
  - cosphi
  - frequency
  - pconsumecounter
  - psupplycounter
THRESHOLD: 10
RECONNECT_INTERVAL: 86400
DEBUG: 0

The Mqtt shows offline

kellerza, 2021-08-05:
Please add the log as well, surround it with three backticks, e.g.

` ``` `
...paste log here
` ``` `

You can also change DEBUG to 1

Note: Your HS core seems 2 months behind and 0.7 is also an older version (although 0.7 was working well)

Jogiarea, 2021-08-05:
Hi Thanks for your fast answer

oh sorry forgotten```

I updated HS Core to core-2021.8.1

And from 0.7 to 2021.8.0 

Changed into Debug : 1

```
ic| sen.name: 'psupply', sen.value: 491.3, delta: -20
ic| topic: 'SMA-EM/status/1900208891/psupply', payload: 491.3
ic| sen.name: 'ssupply', sen.value: 570.7, delta: -20
ic| topic: 'SMA-EM/status/1900208891/ssupply', payload: 570.7
ic| now: 1628155352, val: 505.0, sen.value: 491.3, len(sen.values): 1
ic| topic: 'SMA-EM/status/1900208891/psupply', payload: 488.2
ic| now: 1628155352, val: 583.9, sen.value: 570.7, len(sen.values): 1
ic| topic: 'SMA-EM/status/1900208891/ssupply', payload: 568.8
ic| sen.name: 'psupply', sen.value: 505.0, delta: 10
ic| topic: 'SMA-EM/status/1900208891/psupply', payload: 505.0
ic| sen.name: 'ssupply', sen.value: 583.9, delta: 10
ic| topic: 'SMA-EM/status/1900208891/ssupply', payload: 583.9
ic| now: 1628155353
    val: 64218.0559
    sen.value: 64218.1
    len(sen.values): 60
ic| topic: 'SMA-EM/status/1900208891/pconsumecounter_max'
    payload: 64218.1
ic| now: 1628155353
    val: 227.749
    sen.value: 229.4
    len(sen.values): 60
ic| topic: 'SMA-EM/status/1900208891/u1', payload: 228.3
ic| now: 1628155353, val: 0.868, sen.value: 0.7, len(sen.values): 60
ic| topic: 'SMA-EM/status/1900208891/cosphi', payload: 0.8
ic| now: 1628155353
    val: 64218.0559
    sen.value: 64218.1
    len(sen.values): 60
ic| topic: 'SMA-EM/status/1900208891/pconsumecounter'
    payload: 64218.1
ic| now: 1628155353
    val: 44351.7923
    sen.value: 44351.8
    len(sen.values): 60
ic| topic: 'SMA-EM/status/1900208891/psupplycounter', payload: 44351.8
ic| now: 1628155356, val: 475.4, sen.value: 505.0, len(sen.values): 3
ic| topic: 'SMA-EM/status/1900208891/psupply', payload: 500.6
```

But it dosent work

kellerza, 2021-08-05:
Your debug seems good, but there was a nasty bug with the add-on in `2021.8.0` (I incorrectly interpreted the new energy device classes in homeassistant 

Please update again

To force checking new versions follow this: https://github.com/kellerza/hassio-sma-em#updating-to-a-newer-version

Jogiarea, 2021-08-05:
ok many thanks it works like at the first day

closed 2021-08-05 by Jogiarea