kellerza/hassio-sma-em #8: 2021.8.1: no MQTT communication

northalpha, 2021-08-05:
Maybe this is related to #7 but in my setup i am getting zero information out of your addon.

Setup: KVM Host with HASSOS/Supervisor thing on a bridged network.  I can confirm mosquitto broker, user and pass are correct and working with some python paho script.

Inside the homeassistant terminal as well as inside your container i am seeing the multicast traffic:

```
➜  ~ docker exec -ti addon_6a2c1c7a_sma-em /bin/sh

/usr/src/app # tcpdump -s0 -vvvvv host 239.12.255.254
tcpdump: listening on enp0s3, link-type EN10MB (Ethernet), snapshot length 262144 bytes
15:38:43.195123 IP (tos 0x0, ttl 1, id 50230, offset 0, flags [DF], proto UDP (17), length 636) SMA300xxxxxxx.46883 > 239.12.255.254.9522: [udp sum ok] UDP, length 608
15:38:43.394567 IP (tos 0x0, ttl 1, id 50239, offset 0, flags [DF], proto UDP (17), length 636) SMA300xxxxxxx.46883 > 239.12.255.254.9522: [udp sum ok] UDP, length 608
15:38:43.594985 IP (tos 0x0, ttl 1, id 50250, offset 0, flags [DF], proto UDP (17), length 636) SMA300xxxxxxx.46883 > 239.12.255.254.9522: [udp sum ok] UDP, length 608
15:38:43.794439 IP (tos 0x0, ttl 1, id 50257, offset 0, flags [DF], proto UDP (17), length 636) SMA300xxxxxxx.46883 > 239.12.255.254.9522: [udp sum ok] UDP, length 608
15:38:43.994454 IP (tos 0x0, ttl 1, id 50277, offset 0, flags [DF], proto UDP (17), length 636) SMA300xxxxxxx.46883 > 239.12.255.254.9522: [udp sum ok] UDP, length 608
```

**Addon Config**:

```
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt-user
MQTT_PASSWORD: xxxxx
SMA_SERIALS: []
FIELDS:
  - pconsume
  - pconsumecounter:max
  - u1
  - psupply
  - sconsume
  - ssupply
  - cosphi
  - frequency
  - pconsumecounter
  - psupplycounter
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: 1
AUTODISCOVER_CONFIG:
  - key: state_class
    value: measurement
  - key: last_reset
    value: '2021-07-30T00:00:00+00:00'
```
DNS lookup inside container is working:

```
/usr/src/app # nslookup core-mosquitto
Server:		172.30.32.3
Address:	172.30.32.3#53

Name:	core-mosquitto.local.hass.io
Address: 172.30.33.0
```

**Addon Log**

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
DEBUG:asyncio:Datagram endpoint remote_addr=None created: (<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>, <__main__.MulticastServerProtocol object at 0x7f7b68e96490>)
```

Any idea? Would really love to see this fly :D

kellerza, 2021-08-12:
Hi @northalpha this might be more related to #5 - can you try the legacy addon?

It uses SMA-EM directly for a connection

northalpha, 2021-08-12:
Nice, we are one step further to getting this working. Using the legacy `0.5.5` i get:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/pconsume/config'
    payload: ('{"name": "pconsume", "dev_cla": "power", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/pconsume", "unit_of_meas": "W", "uniq_id": '
              '"300xxxxxxxxx_pconsume", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/pconsumecounter/config'
    payload: ('{"name": "pconsumecounter", "dev_cla": "energy", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/pconsumecounter", "unit_of_meas": "kWh", '
              '"uniq_id": "300xxxxxxxxx_pconsumecounter", "dev": {"ids": '
              '["sma_em_300xxxxxxxxx"], "name": "SMA Energy Meter", "mdl": "Energy Meter", '
              '"mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/u1/config'
    payload: ('{"name": "u1", "dev_cla": "voltage", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/u1", "unit_of_meas": "V", "uniq_id": '
              '"300xxxxxxxxx_u1", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA Energy '
              'Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/psupply/config'
    payload: ('{"name": "psupply", "dev_cla": "power", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/psupply", "unit_of_meas": "W", "uniq_id": '
              '"300xxxxxxxxx_psupply", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/sconsume/config'
    payload: ('{"name": "sconsume", "dev_cla": "energy", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/sconsume", "unit_of_meas": "", "uniq_id": '
              '"300xxxxxxxxx_sconsume", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/ssupply/config'
    payload: ('{"name": "ssupply", "dev_cla": "energy", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/ssupply", "unit_of_meas": "", "uniq_id": '
              '"300xxxxxxxxx_ssupply", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/cosphi/config'
    payload: ('{"name": "cosphi", "dev_cla": "energy", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/cosphi", "unit_of_meas": "", "uniq_id": '
              '"300xxxxxxxxx_cosphi", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/frequency/config'
    payload: ('{"name": "frequency", "dev_cla": null, "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/frequency", "unit_of_meas": "Hz", "uniq_id": '
              '"300xxxxxxxxx_frequency", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
ic| topic: 'homeassistant/sensor/300xxxxxxxxx/psupplycounter/config'
    payload: ('{"name": "psupplycounter", "dev_cla": "energy", "stat_t": '
              '"SMA-EM/status/300xxxxxxxxx/psupplycounter", "unit_of_meas": "kWh", "uniq_id": '
              '"300xxxxxxxxx_psupplycounter", "dev": {"ids": ["sma_em_300xxxxxxxxx"], "name": '
              '"SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
```

So far, nice! But these Sensors are in "unknown" state, but i have found this in HA logs:

```
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 9:41:04 (3 occurrences)
Last logged: 9:44:02

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'frequency', 'device_class': None, 'state_topic': 'SMA-EM/status/300xxxxxxxxx/frequency', 'unit_of_measurement': 'Hz', 'unique_id': '300xxxxxxxxx_frequency', 'device': {'identifiers': ['sma_em_300xxxxxxxxx'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: value must be one of ['battery', 'carbon_dioxide', 'carbon_monoxide', 'current', 'energy', 'humidity', 'illuminance', 'monetary', 'power', 'power_factor', 'pressure', 'signal_strength', 'temperature', 'timestamp', 'voltage'] for dictionary value @ data['device_class']
```

kellerza, 2021-08-16:
`device_class` was never implemented for frequency it seems, from your log:

```
"dev_cla": null
```

kellerza, 2021-08-16:
Try removing frequency for now

northalpha, 2021-08-16:
Unfortunately same "error" (no values seems to be read or feed into mqtt. I am only using `pconsume` in config at this point, I can see that the topic is being created in mosquitto, but no joy until now :(

Addon Log
```
ic| topic: 'homeassistant/sensor/300xxxxx/pconsume/config'
    payload: ('{"name": "pconsume", "dev_cla": "power", "stat_t": '
              '"SMA-EM/status/300xxxxx/pconsume", "unit_of_meas": "W", "uniq_id": '
              '"300xxxxx_pconsume", "dev": {"ids": ["sma_em_300xxxxx"], "name": "SMA '
              'Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}')
```

Inside Mosquitto:
```
{"name": "pconsume", "dev_cla": "power", "stat_t": "SMA-EM/status/300xxxxx/pconsume", "unit_of_meas": "W", "uniq_id": "300xxxxx/_pconsume", "dev": {"ids": ["sma_em_300xxxxx/"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}
```

kellerza, 2021-08-16:
It works ok for me

Do you see any new errors in HomeAssistant? (You might have to restart HASS/delete the persistent entry in MQTT and restart the addon)

If you use MQTTexplorer/something similar do you see anything posted for the value? `SMA-EM/status/300xxxxx/pconsume`?

kellerza, 2021-09-12:
I tried to copy all the connection settings from the ‘legacy’ to the current version

if you still not see values I’ll post some instructions to help debug this

kellerza, 2021-09-21:
@northalpha have you tried the normal version recently?

kellerza, 2022-01-28:
Let me know if you still have an issue

closed 2022-01-28 by kellerza