kellerza/hassio-sma-em #9: Error in Log

johnnysmithy852456, 2021-08-15:
Hi 
Have anyone seen this before.

Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: August 12, 2021, 8:44:09 AM (2 occurrences)
Last logged: August 12, 2021, 8:44:09 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'pconsume_5', 'device_class': 'power', 'state_topic': 'SMA-EM/status/1900248759/pconsume_5', 'unit_of_measurement': 'W', 'unique_id': '1900248759_pconsume_5', 'device': {'identifiers': ['sma_em_1900248759'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'state_class': 'energy', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: value must be one of ['measurement'] for dictionary value @ data['state_class']

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'psupply_5', 'device_class': 'power', 'state_topic': 'SMA-EM/status/1900248759/psupply_5', 'unit_of_measurement': 'W', 'unique_id': '1900248759_psupply_5', 'device': {'identifiers': ['sma_em_1900248759'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'state_class': 'energy', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: value must be one of ['measurement'] for dictionary value @ data['state_class']

kellerza, 2021-08-16:
Hi, it seems like you changed the Autodiscover options for the Addon

Try this:

```
AUTODISCOVER_CONFIG:
  - key: state_class
    value: measurement
```

kellerza, 2021-09-03:
From 2021.9.2 this should be empty

```
AUTODISCOVER_CONFIG: []
```

kellerza, 2021-09-21:
@johnnysmithy852456 did this resolve your issue?

closed 2021-09-23 by kellerza

johnnysmithy852456, 2021-11-03:
Wow sorry for the last reply Yes problem solved.