kellerza/hassio-sma-em #11: Async discover fails with "extra keys not allowed @ data['last_reset']"

tom-winkler, 2021-09-11:
Any idea how to resolve or what a root cause could be?
```
2021-09-11 20:16:18 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'u1', 'device_class': 'voltage', 'state_topic': 'SMA-EM/status/1900*****/u1', 'unit_of_measurement': 'V', 'unique_id': '190*****_u1', 'state_class': 'measurement', 'last_reset': '2021-07-30T00:00:00+00:00', 'device': {'identifiers': ['sma_em_190*****'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['last_reset']
```

kellerza, 2021-09-11:
Do you perhaps have last_reset in AUTODISCOVER options for the addon?

tom-winkler, 2021-09-12:
The config does not reveal anything or do you mean a different plugin?
`MQTT_HOST: homeassistant
MQTT_PORT: 1883
MQTT_USERNAME: '!secret mqtt_user'
MQTT_PASSWORD: '!secret mqtt_pwd'
SMA_SERIALS: []
FIELDS:
  - pconsume
  - pconsumecounter:max
  - u1
THRESHOLD: 80
DEBUG: 0`

I'm checking where this could originate from....still puzzled.

`2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/1900251975/pconsume/config: b'{"name": "pconsume", "dev_cla": "power", "stat_t": "SMA-EM/status/1900251975/pconsume", "unit_of_meas": "W", "uniq_id": "1900251975_pconsume", "state_class": "measurement", "last_reset": "2021-07-30T00:00:00", "dev": {"ids": ["sma_em_1900251975"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}'
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'name': 'pconsume', 'device_class': 'power', 'state_topic': 'SMA-EM/status/1900251975/pconsume', 'unit_of_measurement': 'W', 'unique_id': '1900251975_pconsume', 'state_class': 'measurement', 'last_reset': '2021-07-30T00:00:00', 'device': {'identifiers': ['sma_em_1900251975'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'}
2021-09-12 06:30:53 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 1900251975 pconsume
2021-09-12 06:30:53 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'pconsume', 'device_class': 'power', 'state_topic': 'SMA-EM/status/1900251975/pconsume', 'unit_of_measurement': 'W', 'unique_id': '1900251975_pconsume', 'state_class': 'measurement', 'last_reset': '2021-07-30T00:00:00', 'device': {'identifiers': ['sma_em_1900251975'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['last_reset']

2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('sensor', '1900251975 pconsume'): deque([])
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/1900251975/pconsumecounter_max/config: b'{"name": "pconsumecounter_max", "dev_cla": "energy", "stat_t": "SMA-EM/status/1900251975/pconsumecounter_max", "unit_of_meas": "kWh", "uniq_id": "1900251975_pconsumecounter_max", "state_class": "measurement", "last_reset": "2021-07-30T00:00:00", "dev": {"ids": ["sma_em_1900251975"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}'
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/1900251975/u1/config: b'{"name": "u1", "dev_cla": "voltage", "stat_t": "SMA-EM/status/1900251975/u1", "unit_of_meas": "V", "uniq_id": "1900251975_u1", "state_class": "measurement", "last_reset": "2021-07-30T00:00:00", "dev": {"ids": ["sma_em_1900251975"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}'
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'name': 'pconsumecounter_max', 'device_class': 'energy', 'state_topic': 'SMA-EM/status/1900251975/pconsumecounter_max', 'unit_of_measurement': 'kWh', 'unique_id': '1900251975_pconsumecounter_max', 'state_class': 'measurement', 'last_reset': '2021-07-30T00:00:00', 'device': {'identifiers': ['sma_em_1900251975'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'}
2021-09-12 06:30:53 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 1900251975 pconsumecounter_max
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'name': 'u1', 'device_class': 'voltage', 'state_topic': 'SMA-EM/status/1900251975/u1', 'unit_of_measurement': 'V', 'unique_id': '1900251975_u1', 'state_class': 'measurement', 'last_reset': '2021-07-30T00:00:00', 'device': {'identifiers': ['sma_em_1900251975'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufact [... cut ...]
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['last_reset']

2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('sensor', '1900251975 pconsumecounter_max'): deque([])
2021-09-12 06:30:53 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('sensor', '1900251975 u1'): deque([])
`

tom-winkler, 2021-09-12:
Looks like I comeplety need to mit the last_rest attribute to get it working.

`2021-09-12 08:04:15 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on homeassistant/sensor/1900251975/pconsume/config: '{"name": "pconsume", "dev_cla": "power", "stat_t": "SMA-EM/status/1900251975/pconsume", "unit_of_meas": "W", "uniq_id": "1900251975_pconsume", "state_class": "measurement", "dev": {"ids": ["sma_em_1900251975"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}', mid: 34
2021-09-12 08:04:15 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/1900251975/pconsume/config: b'{"name": "pconsume", "dev_cla": "power", "stat_t": "SMA-EM/status/1900251975/pconsume", "unit_of_meas": "W", "uniq_id": "1900251975_pconsume", "state_class": "measurement", "dev": {"ids": ["sma_em_1900251975"], "name": "SMA Energy Meter", "mdl": "Energy Meter", "mf": "SMA"}}'
2021-09-12 08:04:15 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'name': 'pconsume', 'device_class': 'power', 'state_topic': 'SMA-EM/status/1900251975/pconsume', 'unit_of_measurement': 'W', 'unique_id': '1900251975_pconsume', 'state_class': 'measurement', 'device': {'identifiers': ['sma_em_1900251975'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'platform': 'mqtt'}
2021-09-12 08:04:15 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 1900251975 pconsume
2021-09-12 08:04:15 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.pconsume`

kellerza, 2021-09-12:
Thank you for the PR @tom-winkler , I see this change was on the -dev version

Please check that you are not using the development version of this addon

closed 2021-09-12 by kellerza