kellerza/hassio-sma-em #15: Failed to install add-on

seansassen, 2021-11-14:
Hi there, 

Getting the following error when trying to install the add-on, any guidance would be appreciated.

Thanks.

_**Failed to install add-on**

The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir paho-mqtt~=1.5.0 attrs requests~=2.24.0 icecream' returned a non-zero code: 1_

seansassen, 2021-11-16:
@kellerza any idea?

PS. Also from RSA :)

kellerza, 2021-11-16:
Hey @seansassen sorry was a bit preoccupied mountain biking over your mountains this weekend.

The error does not say much, but likely just connectivity/DNS related. Do you have a static IP on your hass os? Do you use adguard/pihole etc?

seansassen, 2021-11-16:
No problem, that's obviously a more important hobby.

I do have a static ip, and I did have adguard installed - just tried to deactivate it, but still got the same error.

kellerza, 2021-11-16:
Can you check the "supervisors" logs and see if you get any more information? What version of the supervisor are you running? "stable channel"?

In [this](https://community.home-assistant.io/t/sma-energy-meter-in-home-assistant/47482/156) chat someone had a similar problem, but was DNS related. That might also be a good place to check

seansassen, 2021-11-16:
`Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError:
[Errno 1] Operation not permitted`

seansassen, 2021-11-16:
Running supervisor-2021.10.8, but will look into the other chat and see if anything I can find.

Thanks for all your help.

kellerza, 2021-11-16:
From the community there is also this - https://community.home-assistant.io/t/2021-7-a-new-entity-trigger-ids-and-script-debugging/320556/50?u=kellerza - expand the "Replies" on this topic

seansassen, 2021-11-16:
All resolved, thanks for your help @kellerza

closed 2021-11-16 by seansassen