kellerza/hassio-sma-em #19: Home Assistant auto-discovery | Can't get it to work

prom00, 2022-08-01:
I'm having issues setting up the add-on.

I've installed:
Mosquitto broker
Current version: 6.1.2

I'm new to MQTT, but when using MQTT explorer I can connect to the MQQT :
[image]

Those are my settings:
[image]

Looking at the ingegrations screen in HA, I didn't see the MQTT, I've added it manually:
[image]

I'm missing any entities auto discovered? 

I'm running HA:
[image]

In the SMA-EM config I've got this:
[image]

`MQTT_HOST: 192.168.1.50
MQTT_PORT: 1883
MQTT_USERNAME: hassmqqt`

In HA config/users:
[image]

I'm not sure what I'm missing here, but to me it looks like the data isn't going from the SMA-EM to MQTT?

prom00, 2022-08-01:
Thanks alot for your fast reply.

I've got this in my logs:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.

I also got this logs in Mosquitto:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] mosquitto.sh: executing... 
[09:29:58] INFO: SSL is not enabled
[cont-init.d] mosquitto.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[09:29:58] INFO: Starting NGINX for authentication handling...
[09:29:58] INFO: Starting mosquitto MQTT broker...
1659338998: Warning: Mosquitto should not be run as root/administrator.
[09:29:59] INFO: Successfully send discovery information to Home Assistant.
[09:29:59] INFO: Successfully send service information to the Supervisor.

kellerza, 2022-08-01:
It seems like you never receive any multicast traffic from the SMA-EM. Once you receive your first reading the discovery process will start, see the bold lines below from my log

<pre>
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
WARNING:asyncio:Executing Handle _SelectorDatagramTransport._read_ready() created at /usr/local/lib/python3.9/asyncio/selector_events.py:259 took 0.128 seconds
<b>INFO:sensors:Discover 9/9 sensors on SMA 19xxxxx
INFO:sensors: - u1 every 60s</b>
INFO:mqtt:Connecting
INFO:mqtt:Connecting
INFO:mqtt:MQTT: Connection successful
INFO:sensors: - u1_min every 60s
INFO:sensors: - u1_max every 60s
INFO:sensors: - p1consume every 60s
INFO:sensors: - p1consumecounter_max every 60s
INFO:sensors: - p2consume every 60s
INFO:sensors: - p2consumecounter_max every 60s
INFO:sensors: - p3consume every 60s
INFO:sensors: - p3consumecounter_max every 60s
</pre>

Is everything wired up and does your home network support IGMPv2?

prom00, 2022-08-01:
What do you mean with 'Wired up'?

I've enabled debug (again). I had it on before, but tried reinstalling everything.

[image]

I'm not sure whether my home network supports IGMPv2. But I'm just running a basic network with a couple of routers & switches.

I'm running HA on a Truenas server inside a VM. On HA is also MQTT installed so they should be able to talk locally right?

prom00, 2022-08-01:
I've tried sending messages through MQTT with MQTTX program.
Setup 2 interfaces and send a message on one and it was received by the other.
[image]

I'll take it that MQTT is working.

kellerza, 2022-08-01:
You don’t get anything from the energy meter… (mqtt is not involved yet)

https://github.com/kellerza/hassio-sma-em/issues/5#issuecomment-944697605

prom00, 2022-08-01:
https://github.com/datenschuft/SMA-EM

Do I need to install the above manually?

kellerza, 2022-08-25:
@prom00 everything that is required is already added, but if the multicast / IGMPv2 packets never arrive, then there is no way of understanding this.
See if you can remove switches/routers in your network and have a more direct line between the EM and you Hass instance to test this theory,- https://github.com/kellerza/hassio-sma-em/issues/5#issuecomment-944697605

kellerza, 2022-08-25:
The addon waits for the first valid multiast packet before it will attempt to make any connection to the MQTT server

prom00, 2022-08-31:
I've got this working with another method now. I'm using MODBUS:

```
modbus: 
  - type: tcp
    host: 192.168.1.151 # ip nummer omvormer
    port: 502
    name: "sma"
    close_comm_on_error: true
    delay: 5
    timeout: 5
    sensors:
      - name: SMA_inverter_energy_day # Totaal opgewekt deze dag (enige echte belangrijke)
        unique_id: 12345PV
        unit_of_measurement: kWh
        slave: 3
        address: 30517 # Daily yield (Wh) [E-heute]
        count: 4
        scale: 0.001
        state_class: total_increasing
        device_class: energy
        precision: 2
        data_type: int64
```

I'm not sure if this is the same thing you are using? I found out that my inverter doesn't have the ability to be connected through web, but only through modbus.

kellerza, 2022-08-31:
Are you sure you are getting this from your SMA-Energy Meter? (SMA-EM)

It seem like you are rather reading information from an Inverter with your config snippet?

- This addon is for the SMA-EM only, it receives data through multicast - https://github.com/kellerza/hassio-sma-em
- There is also pysma, that uses the WebConnect interface - https://github.com/kellerza/pysma (this is integrated into Home Assistant) - but will not work with Modbus

dirkmueller2, 2023-09-27:
After a lot of reading and trying out I found a post on another board.
Its in German: https://community.simon42.com/t/sma-home-manager-auslesen/4560/20

Solution for me, now its works perfect, is to write down the IP from the Home Assitant Server into the Homemanager Portal in the Zählerkonfiguration.

Its on the settings of the Home Manager in SunnyPortal:

[image]

[image]

The IP of the HomeAssitant Server should be not changed from the router, thats fine on my config in the Fritzbox. No its working perfect. Hard job but no I am Happy.

[image]

Tomi05-k, 2024-01-08:
Wichtig zu ergänzen ist, dass man oben bei Ansicht auf Erweiterte Konfiguration geht. Sonst wird der Punkt nicht angezeigt und man sucht wie ich 2h danach. 
[image]

Viele Grüße Tom!