kellerza/hassio-sma-em #20: consume and consumecounter have wrong device_class

gabrio79, 2022-08-24:
sensor.s1consume has unit VA which is unsupported for device_class energy ---> must have device_class: apparent_power
sensor.s3consume has unit VA which is unsupported for device_class energy
sensor.s2consume has unit VA which is unsupported for device_class energy
sensor.s2consumecounter has unit kVAh which is unsupported for device_class energy
sensor.s3consumecounter has unit kVAh which is unsupported for device_class energy

kellerza, 2022-08-25:
Thanks @gabrio79! 
Any idea if there is a device_class that could match `kVah`, or simply no device class / `None`?

gabrio79, 2022-08-25:
I searched but i dis't applicabile class foundee, i can suggest you to open
an issue on ha-core? in the midtime, no class or remain energy

gabrio79, 2022-08-25:
Is you device accurate? I notated big differces betwen shm2 and fiscale
Meter measure... I opened a ticket on sma support... But for them no
issue... Have you ever checked?

alexfiftyfour, 2023-11-07:
This is the device class list.
https://www.home-assistant.io/integrations/sensor/

I have a similar problem with the current.

kellerza, 2023-11-08:
Please try the -dev add onhttps://github.com/kellerza/hassio-sma-em/blob/main/sma-em-dev/Dockerfile#L4

closed 2024-02-06 by kellerza