kellerza/hassio-sma-em #21: Resolution of counter Variables

littleyoda, 2022-10-24:
Is it possible to increase the resolution of the "counter" variables?
Currently only two decimal places are passed to HA. It would be nice if all 4 decimal places are passed to HA.

kellerza, 2022-11-09:
Ok, will look into this when working on this again

Or a PR is also welcome 😉

littleyoda, 2022-11-09:
1.) What would you prefer?
Make the number of decimal places configurable or set it to 3?
I would tend to the variant with 3 fixed decimal places, since the output can be adjusted in HA.

2.) Do I need to change it to both sma-em and sma-em-dev?

kellerza, 2023-05-18:
Added your suggestion here - https://github.com/kellerza/hassio-sma-em/commit/4df12f3b385b9196c44c566ad88abfd64773f37e

closed 2023-05-18 by kellerza