kellerza/hassio-sma-em #22: My SMA-meter is not being found

RobertoCravallo, 2022-11-05:
When I first installed this, thank you by the way, it "found" my HM2 without a hitch.

Then, for some unknown reason, not anymore. Deinstalling or installing the other two flavors changed nothing. Only my inverter is found and that is no help.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
INFO:sensors:Unknown sensor: pconsume
INFO:sensors:Unknown sensor: pconsumecounter
INFO:sensors:Unknown sensor: u1
INFO:sensors:Discover 0/3 sensors on SMA 42xxxxx295

The strange part is, my inverter (SUNNY BOY 5.0) has NEVER found HM nor HM2. All SMA Software works though. PORTAL, PLACES and the ANDROID App all deliver reliable data.

I successfully built a GLOW, but that cannot read the power to grid. GRID_POWER is always positiv, even when supplying to the grid.

Why could this be happening. I am assuming all cables are ok, as the SMA software works ok. Is it possible, that there is a problem with cables if SMA can read all data?

Thanks for your troubles,

Regards from Mannheim, Germany

RobertoCravallo, 2022-11-06:
MQTT Broker does see my speedwire, still SMA-Meter is not finding it.

[image]

kellerza, 2022-11-07:
@RobertoCravallo what version of the addon are you using?

[image]

Not sure why your shows "Unknown sensor" for pconsume & pconsumecounter !?
So it shows you did not configure any known sensor: **0**/3

As an example, with the following config
```yaml
FIELDS:
  - u1
  - u1:min
  - u1:max
  - p1consume
  - p1consumecounter:max
  - p2consume
  - p2consumecounter:max
  - p3consume
  - p3consumecounter:max
  - pconsume
  - pwrong
```
 
I get this
```log
INFO:sensors:Unknown sensor: pwrong
INFO:sensors:Discover 10/11 sensors on SMA 19xxxx00
INFO:sensors: - u1 every 60s
INFO:mqtt:Connecting
INFO:mqtt:MQTT: Connection successful
INFO:sensors: - u1_min every 60s
INFO:sensors: - u1_max every 60s
INFO:sensors: - p1consume every 60s
INFO:sensors: - p1consumecounter_max every 60s
INFO:sensors: - p2consume every 60s
INFO:sensors: - p2consumecounter_max every 60s
INFO:sensors: - p3consume every 60s
INFO:sensors: - p3consumecounter_max every 60s
INFO:sensors: - pconsume every 60s
```

gabrio79, 2022-11-07:
I don't know if is same issue, but since 2 days has stopped works
```
Log Details (ERROR)
Logger: homeassistant.util.logging
Source: util/logging.py:156
First occurred: November 4, 2022 at 11:00:37 PM (2 occurrences)
Last logged: 7:26:50 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'speedwire-version', 'device_class': 'energy', 'state_topic': 'SMA-EM/status/3002136668/speedwire-version', 'unit_of_measurement': None, 'unique_id': '3002136668_speedwire-version', 'device': {'identifiers': ['sma_em_3002136668'], 'name': 'SMA Energy Meter', 'model': 'Energy Meter', 'manufacturer': 'SMA'}, 'state_class': 'total_increasing', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 327, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string value is None for dictionary value @ data['unit_of_measurement']
```
could be something regarding new core release?

kellerza, 2022-11-07:
Do you use the **speedwire-version** sensor? I've never used it but also see the error if I add it

It needs a "unit_of_measurement"...

kellerza, 2022-11-07:
@RobertoCravallo please also try removing **speedwire-version**

RobertoCravallo, 2022-11-07:
I did not knowingly install the speedwire sensor, I will have to hunt it down. I am pretty new to this, so please bare with me. The good news is, at least the HM2 is being detected, so that's a start.

Also I was not clear on the MQQT-Username and password. Do I need to use the mosquitto credentials or the ones from the mqqt-user I configured? Tried both with the same results.

Just an update I will go hunting for the speedwire-version in a bit and let you know. Thank you so far

[image]

[image]

gabrio79, 2022-11-07:
for me, some integration reboot, had resolved
:-)

RobertoCravallo, 2022-11-07:
I have rebooted a lot!!!  :-( No change.

Can't find out, where this speedwire-sensor is coming from....
[image]
[image]

RobertoCravallo, 2022-11-07:
I have the SMA integration installed, no speedwire there! Where the heck is that coming from?
[image]

RobertoCravallo, 2022-11-07:
I am getting tons of this...

[image]

RobertoCravallo, 2022-11-07:
Now, I am totally confused!!! After deinstalling EVERYTHING absolutely not needed, nothing changed. So back to the old GOOGLE. Found the tip to install [SunSpec](https://community.home-assistant.io/t/sma-solar-integration/287602/55), so what the heck, tried everything else I could think of. 

Only installed it via HACKS could not add as integration, IP and port where rejected. But all of a sudden I am noticing, that MQTT EXPLORER is going wild. The data seems to be coming in, even though the SMA-energy meter integration is still showing errors.

Another thing happened: My INVERTER suddenly actually "finds" the HM2. I have ALWAYS had a red "Gerätestatus" for the HM2. Not anymore. This is very, very strange. Could the installation of SupSpec have touched something of? Sure seems so. I deinstalled SunSpec, and it is still working. I will reboot and see what happens.

Now to figure out, what is supplying SMA-EM data. Computers are just nuts sometimes!!!
[image]

[image]

[image]

RobertoCravallo, 2022-11-08:
Now I know what the magic cure is!! Found it on a german website and they had the tip from a german solar-forum.

With all the software deinstalling and generally mucking around, I completely forgot one very important thing.

It was suggested to put all SMA on a switch, with nothing else. I had a bunch of stuff besides the SMA parts on my switch. This "SMA-Switch" is then plugged into the switch of your choice, and that is it. Worked like a charm and got all the SMA gremlins sorted. As soon as I put everything back on a single switch, my inverter can't find the HM2 anymore.

Best part, I had a cheapo [switch](https://www.amazon.de/gp/product/B000FNFSPY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) lying around and it worked out just perfect. 

So to all of you out there have some sort of gremlins in your SMA setup, try this very cheap fix. Worked perfectly for me, "Gerätestatus" is still green. Changes back to red, if everything goes on one switch!!!

[image]

kellerza, 2022-11-09:
Glad you got it working as per #23 

the speedwire-version you see in Mqtt explorer might be from an earlier run of the addon. I do not think the addon cleans any sensors (only started adding that in my sunsynk addon). So that means the only way to remove a sensor from mqtt discovery would be to delete that key with a tool like mqtt explorer

RobertoCravallo, 2022-11-09:
I would assume that a new, much more expensive switch would be "better" that a cheapo plastic one. 
No matter it is working and now I get all data with the SMA-integration and don't need the sma-em integration to figure usages! Perfect.  

For me WIN WIN, thank you for your time and effort.

closed 2022-11-09 by kellerza

dirkmueller2, 2023-09-27:
After a lot of reading and trying out I found a post on another board.
Its in German: https://community.simon42.com/t/sma-home-manager-auslesen/4560/20

Solution for me, now its works perfect, is to write down the IP from the Home Assitant Server into the Homemanager Portal in the Zählerkonfiguration. 

[image]

The IP of the HomeAssitant Server should be not changed from the router, thats fine on my config in the Fritzbox. No its working perfect. Hard job but no I am Happy. 

[image]

[image]