kellerza/hassio-sma-em #23: SMA and METER problem solved!!!!!

RobertoCravallo, 2022-11-08:
Now I know what the magic cure is!! Found it on a german website and they had the tip from a german solar-forum.

With all the software deinstalling and generally mucking around, I completely forgot one very important thing.

It was suggested to put all SMA on ONE switch, with NOTHING else. I had a bunch of stuff besides the SMA parts on my switch. This "SMA-Switch" is then plugged into the switch of your choice, and that is it. Worked like a charm and got all the SMA gremlins sorted. As soon as I put everything back on a single switch, my inverter can't find the HM2 anymore.

Best part, I had a cheapo [switch](https://www.amazon.de/gp/product/B000FNFSPY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) lying around and it worked out just perfect. Was exactly the one recommended to fix this!!! Go figure...

So to all of you out there that have some sort of gremlins in your SMA setup, try this very cheap fix. Worked perfectly for me, "Gerätestatus" is still green. Changes back to red, if everything goes on one switch!!!

[image]

kellerza, 2022-11-09:
Great news!! Some of the comms (especially the EM) requires that the switch supports IGMPv2.

So either your switch supports it, or simply forward these over all ports, bit glad you got it working!!

closed 2022-11-09 by kellerza

RobertoCravallo, 2022-11-09:
I would assume that a new, much more expensive switch would be "better" that a cheapo plastic one. 
No matter it is working and now I get all data with the SMA-integration and don't need the sma-em integration to figure usages! Perfect.  

For me WIN WIN, thank you for your time and effort.