kellerza/hassio-sma-em #25: Can't edit fields

Wackeljupp, 2023-02-02:
Got that running now.
But it only "polls" pcconsume, pconsumecounter:max and u1.
If i try to ad other sensors as like psupply the entry just dissapears.
Am I doing something wrong?
[image]

kellerza, 2023-02-03:
Try adding them in yaml mode?

use the three dots on the top right to switch

Wackeljupp, 2023-02-03:
Okay. That worked.
When i enter the serial of my homemanger2.0 i can't safe it:
Fehler beim Speichern der Add-on Konfiguration, Invalid list for option 'SMA_SERIALS' in SMA Energy Meter

An which fields i have to use to get the values of "PV-Erzeugung", "Gesamtverbrauch" und "Netzbezug/Netzeinspeisung"?
An how and which values/fields i have to use to get the engery dashboard working?

kellerza, 2023-02-06:
It depends how your meter is wired...

The nergy dashboard requires one of the counters (kWh).... if these sensor shave all the correct attributes I'm not sure you might have to manually customise them

kellerza, 2023-05-18:
If the serial is a number you should wrap it in quotes to make it a string

closed 2023-05-18 by kellerza