kellerza/hassio-sma-em #26: No sensors in the MQTT

Benleni17, 2023-02-08:
Good evening community, 

I am not receiving any sensors in the MQTT. What am I doing wrong? Attached is the log.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
DEBUG:asyncio:Datagram endpoint remote_addr=None created: (<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>, <__main__.MulticastServerProtocol object at 0x7fad84d3a0>)
INFO:sensors:Unknown sensor: pconsume
INFO:sensors:Unknown sensor: pconsumecounter
INFO:sensors:Unknown sensor: u1
INFO:sensors:Unknown sensor: frequency
INFO:sensors:Unknown sensor: p2consume
INFO:sensors:Unknown sensor: p2supply
INFO:sensors:Unknown sensor: p3consume
INFO:sensors:Unknown sensor: p3supply
INFO:sensors:Unknown sensor: q3consume
INFO:sensors:Unknown sensor: speedwire-version
INFO:sensors:Unknown sensor: pconsumecounter
ic| res: []
INFO:sensors:Discover 0/11 sensors on SMA 4294967295
WARNING:asyncio:Executing <Task pending name='Task-2' coro=<process_emparts() running at /usr/src/app/sensors.py:85> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x7fad6bc070>()] created at /usr/local/lib/python3.9/asyncio/base_events.py:424> created at /usr/src/app/./run.py:29> took 0.299 seconds
INFO:sensors:Unknown sensor: pconsume
INFO:sensors:Unknown sensor: pconsumecounter
INFO:sensors:Unknown sensor: u1
INFO:sensors:Unknown sensor: frequency
INFO:sensors:Unknown sensor: p2consume
INFO:sensors:Unknown sensor: p2supply
INFO:sensors:Unknown sensor: p3consume
INFO:sensors:Unknown sensor: p3supply
INFO:sensors:Unknown sensor: q3consume
INFO:sensors:Unknown sensor: speedwire-version
INFO:sensors:Unknown sensor: pconsumecounter
ic| res: []
INFO:sensors:Discover 0/11 sensors on SMA 4294967295

Greetz
Benedikt

dominikruck1, 2023-03-19:
@Benleni17 did you solve the probem or did you connect the SMA in other way

hjschwe, 2023-05-12:
@Benleni17    Same Error. It is strange that you have the same SMA system detected in your log file ??
....
INFO:sensors:Unknown sensor: pconsume
INFO:sensors:Unknown sensor: pconsumecounter
INFO:sensors:Unknown sensor: u1
INFO:sensors:Discover 0/3 sensors on SMA 4294967295

I have installed the Dev version and get new entries in log file:

[services.d] done.
2023-05-12 09:31:36,005 INFO    options: Loading HASS OS configuration
2023-05-12 09:31:36,006 INFO    MQTT: Connecting to mqtt@core-mosquitto:1883
2023-05-12 09:31:36,009 INFO    MQTT: Connection successful
2023-05-12 09:31:36,518 INFO    Listening for multicast frames. Sensor discovery will be triggered by the first frame.
2023-05-12 09:31:51,103 INFO    Ignore protocol 0x6065
2023-05-12 09:32:25,586 INFO    Ignore protocol 0x6065
2023-05-12 09:32:56,117 INFO    Ignore protocol 0x6065

Results:  No data detected.

kellerza, 2023-05-18:
This seems related to #27 and #28 (fixed in dev)

- What versions of the addon are you running?
- What SMA HW do you have on your network?

hjschwe, 2023-06-04:
@kellerza - now works OK for me. Receiving the datas. No further problems detected.

Installed dev version is:   2023.5.18
SMA Smart Meter SHM2.0

closed 2023-06-04 by kellerza