kellerza/hassio-sma-em #30: Bekomme plötzlich keine Daten mehr.

Bimboboy007, 2023-04-27:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
DEBUG:asyncio:Datagram endpoint remote_addr=None created: (<_SelectorDatagramTransport fd=8 read=idle write=<idle, bufsize=0>>, <__main__.MulticastServerProtocol object at 0x7f5ff90423a0>)

Bimboboy007, 2023-04-27:
Habe jetzt zum testen die developer version installier da bekomme ich wieder Daten aber falsche Werte.

Bimboboy007, 2023-04-27:
Funktioniert wieder hat anscheinend ein Problem mit dem update von Operating System 10.1 gegeben habe backup eingespielt und alles läuft wie gewohnt.

tom-winkler, 2023-09-25:
Danke - selbes problem hier....hoffe ein neues OS behebt es alsbald wieder?

closed 2024-02-12 by kellerza