kellerza/hassio-sma-em #33: Add-On suddenly stopped working

askmike72, 2023-11-11:
Hi,

two days ago it worked perfectly, suddenly I only get this readings:
INFO:sensors:Unknown sensor: pconsume
INFO:sensors:Unknown sensor: psupply
INFO:sensors:Unknown sensor: u1
INFO:sensors:Unknown sensor: u2
INFO:sensors:Unknown sensor: u3
INFO:sensors:Unknown sensor: pconsumecounter
INFO:sensors:Unknown sensor: psupplycounter
INFO:sensors:Discover 0/7 sensors on SMA 4294967295

Even with the serial of my SHM configured.

The SHM works, because I have a OpenWB wallbox running which reads the data correctly.

permezel, 2024-02-05:
Me too.  I was able to get it working, worked great for a few days, then stopped.  uninstall/reinstall has not helped.

kellerza, 2024-02-06:
Usually this is due to some network change/ the multicast frames not coming through.

Any OS upgrades, etc?

permezel, 2024-02-07:
No upgrades.  Running in VM on Synology.  Network has not changed.  I do have dual LANs and the 9522 packets are only on one of them.  This morning, I rebooted the NAS box, and apparently it started working again.  I used to have an iOS app (emview, or something) which was great for figuring out what was going on, but it fell out of the app store as it was not being updated. 

I just checked my `sunnyportal` for the last few days.  No outages on the energy-meter.  My SMA-Home-Manager, residing on the ioT LAN, snoops the 9522 reports from the energy-meter (also on the ioT LAN) and reports to `sunnyportal.com`, so there was not an issue on the lan or the devices.

Next time I see it drop out, I will ensure the Synology is receiving the 9522 pkts (I have no reason to suspect not).  I will also `docker exec` into the relevant location and see if I can tell.  `tcpdump` is not available, IIRC, and `python3` is also curiously absent some times (although I think I fixed that), so I can always write a python UDP/9522 snooper.

Here is the current running instance:
[image]

I guess a UDP socket bound to port 9522 will get packets from all the interfaces, so it should work.  `192.168.4.0/24` is the ioT lan.
If it happens again, I will poke further.

permezel, 2024-02-07:
Turns out it is not currently running.  It did, apparently, re-start running this morning, resulting in the daily history display being partially populated.  Was not running when I was poking about earlier.  So I can confirm that port 9522 is received on the Synology from the energy meter (and from the Home-manager, and from the direct-wired 5000-TL inverter).

I just found `nc` in the docker image.  Not very familiar with that, but `nc -u -v -p 9522 -l -s 239.12.255.255` yields nothing, whereas there is a steady supply of hits via `tcpdump` on the Synology.
Tried `-s 0.0.0.0` as well.  I confirm that there are UDP sockets in listen mode bound in both the `sma-em` addon and in the main VM.  There is no `tcpdump` nor `nc` in the main VM, so I can't see if the packets reach there.  I can ping out to hosts on the `ioT` LAN, specifically to the host sending the 9522 packets.

I just found a `iptables` firewall on the main VM, which at first glance, seems to not allow UDP in, so I flushed it.  Default input policy now accept.  Don't know if main VM firewall is propagated to docker instances, but the `sma-em` instance still does not see any 9522 pkts.

permezel, 2024-02-07:
Just found a HassOs update.  applied it.  Got my sensors back.  Perhaps that was it.

permezel, 2024-02-07:
I note that the `sma-em` add-on does not process the solar supply packets I am getting, just the energy-meter.
Any plans to add sensors for the rest?

permezel, 2024-02-07:
[image]
vs
[image]

kellerza, 2024-02-12:
> I note that the `sma-em` add-on does not process the solar supply packets I am getting, just the energy-meter. Any plans to add sensors for the rest?

No plans to do this. This addon is only for the EM.

pysma has more options, and built into HASS.

closed 2024-02-12 by kellerza