kellerza/hassio-sma-em #34: Warning:  is using native unit of measurement 'A' which is not a valid unit for the device class ('energy')

matze338, 2024-02-09:
I get the following warning message in Homeassistant for the fields i1, i2 and i3:
```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:738
Integration: Sensor (documentation, issues)
First occurred: 10:17:07 (3 occurrences)
Last logged: 10:17:07

Entity sensor.sma_energy_meter_i1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'A' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'Wh', 'GJ', 'MWh', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.sma_energy_meter_i2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'A' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'Wh', 'GJ', 'MWh', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.sma_energy_meter_i3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'A' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'Wh', 'GJ', 'MWh', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```
I guess this entities needs `state_class: measurement` and `device_class: current` instead of `total_increase` and `energy`.

Config:
```
MQTT_HOST: IP
MQTT_PORT: 1883
MQTT_USERNAME: USER
MQTT_PASSWORD: PW
SMA_SERIALS: []
FIELDS:
  - pconsume
  - pconsumecounter
  - psupply
  - psupplycounter
  - frequency
  - p1consume
  - p1consumecounter
  - p1supply
  - p1supplycounter
  - p1consume
  - i1
  - u1
  - p2consume
  - p2consumecounter
  - p2supply
  - p2supplycounter
  - i2
  - u2
  - p3consume
  - p3consumecounter
  - p3supply
  - p3supplycounter
  - i3
  - u3
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: false
AUTODISCOVER_CONFIG: []

```

kellerza, 2024-02-09:
Can you try the `-dev` addon? #20

matze338, 2024-02-09:
Thanks i1, i2 and i3 now works, but

in the dev-Version i get some "weird" decimal numbers
[image]
is this behavior correct?

kellerza, 2024-02-09:
I'm not sure if anything changes wrt rounding (vaguely recall someone wanting 3 decimal points)

But in the latest HASS frontend you can change how many you want displayed, on a per sensor basis

closed 2024-02-09 by matze338