kellerza/hassio-sma-em #37: Configure threshold below current minimum of 10

TheAssassin, 2024-02-12:
I don't quite see where this lower limit is coming from. An average SMA EM does not seem to produce more than one message a second, which in my opinion is a good rate and makes it well usable for purposes like EVCC which probably benefit from a short interval. The `:N` average functions already allow the user to apply a rate limit, and the plugin would send only one message every `N` seconds for those "channels" onto the MQTT bus. This could be used just fine within Home Assistant then. And given that the events are time discrete anyway, there is, at least in my opinion, no reason to implement a different rate limit than a time-based one.

kellerza, 2024-02-12:
Sorry, I'm not really following. If you don't need the filters, just don't use them.

closed 2024-07-08 by kellerza

imitation, 2025-01-08:
Found this thread beacause I ran into the same limitation: hassio-sma-en enforces a minimum threshold of 10, which seems unneccessary and limits the accuracy of any power tracking using this data.

TheAssassin, 2025-02-14:
@kellerza this issue was closed prematurely. Let me try to rephrase.

[image]

What I was trying (and I think @imitation, too) was to configure the "THRESHOLD" setting to a value lower than 10. This should be supported by the SMA device, but the plugin prevents it without any obvious reason.

For instance, I use the value of 10 because I want the higher precision than the default of 80 seconds. I'd actually enjoy an even greater precision because I want to use this smart meter to control, e.g., the charging of a BEV with a software called EVCC to charge any excess energy into the car's battery rather than send it back to the grid.

I think @imitation also wants to use a smaller threshold because they want to use the data generated by the energy meter to control some process.

The value range of 10 to 1000 is enforced by https://github.com/kellerza/hassio-sma-em/blob/main/sma-em/config.yaml as far as I can see.

Can you tell us whether there is a technical reason for this lower range limit? If there is none, could we reduce it to, say, 1? The SMA devices appear to send one message every second by default anyway.

Side note: I assume that threshold is a bit misleading, too...

kellerza, 2025-02-15:
If you add the sensors with `:1` at the end of the name it should report every second, irrespective of the threshold

TheAssassin, 2025-02-15:
The threshold effectively rate limits the MQTT messaging, doesn't it? I would like to reduce the update interval there.

kellerza, 2025-02-15:
If a change in value exceeds the threshold it will be reported immediately.
Else it will only be reported at the end of the time period.

The values change almost with every read, so setting the threshold to 1 or setting the time period to 1s should have a similar effect

The idea of the filter is to report less, so setting threshold to 1 is pointless, rather just not use the filter in the first place

TheAssassin, 2025-02-15:
I see. I suppose I understood this setting incorrectly then. Thanks for the clarification. I suggest that you document the purpose of the configuration settings somewhere.

TheAssassin, 2025-02-15:
Oh, actually, you have not provided a reason not to allow smaller thresholds, too. You just said it was kind of pointless, but that would be acceptable to me. Would you accept a PR?