kellerza/hassio-sma-em #45: Sensors with wrong state class total_increasing

faulesocke85, 2024-08-23:
Hi, mentioned this error in my log with the latest version. Is it possible to change the class by myself? Or has it to be changed in a fix: 

Entity sensor.sma_energy_meter_i3 from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 1.5 (1.6) with last_updated set to 2024-08-23T11:20:15.325234+00:00.

Entity sensor.sma_energy_meter_cosphi from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 0.9 (1.0) with last_updated set to 2024-08-23T17:54:41.317826+00:00. 

BR

kellerza, 2024-08-23:
Which addon? The dev is probably has the best behavior

faulesocke85, 2024-08-23:
This Add-On which I am posting on… I thought ? 

Home Assistant SMA Energy Meter Add-On

kellerza, 2024-08-24:
Try using the **developer version**, it is more up to date