kellerza/hassio-sma-em #47: The entity no longer has a state class

faulesocke85, 2024-11-21:
Hi there,

I've a problem with the sma em entities reporting they have no longer a state class which turns into a long term statistics problem as ha os is reporting. Any thoughts on this? 

[image]
[image]

kellerza, 2025-01-13:
Can you try the latest release?

faulesocke85, 2025-01-13:
I‘ve updatet, Looks like solved. Thank You

faulesocke85, 2025-01-13:
I‘be made the Update and clicked resolve. 
now my history of the entitities before 4th January 2025 are lost.  I am not shure why since 4.1.2025… 
And I am bot Shure if there is a Problem with the Recording in General …

closed 2025-01-15 by kellerza