kellerza/hassio-sma-em #51: Since Update to 2025.1. error autodiscover_config

app-ch, 2025-01-13:
Since Update the MQTT for SMA is not working anymore:

Log Files:

2025-01-13 14:10:02,998 INFO    options: Loading HASS OS configuration
Traceback (most recent call last):
  File "/usr/src/app/run.py", line 89, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 686, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/src/app/run.py", line 58, in main
    init_options()
  File "/usr/src/app/options.py", line 50, in init_options
    OPT.update(loads(hassosf.read_text(encoding="utf-8")))
  File "/usr/src/app/options.py", line 35, in update
    setattr(self, key.lower(), val)
AttributeError: 'Options' object has no attribute 'autodiscover_config'
[14:10:03] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

kellerza, 2025-01-13:
Can you remove autodiscoved_config from your config?

kristian, 2025-01-13:
If I try to remove the empty array `[]` in the YAML and try to save in the Home Assistant UI, I get the error that:

`Fehler beim Speichern der Add-on Konfiguration, Missing option 'AUTODISCOVER_CONFIG' in root in SMA Energy Meter (6a2c1c7a_sma-em). Got {'MQTT_HOST': 'core-mosquitto', 'MQTT_PORT': 1883, 'MQTT_USERNAME': 'mqtt', 'MQTT_PASSWORD': 'fC9n4RDbXNfadTYB', 'IPBIND': '', 'SMA_SERIALS': ['3015984927'], 'FIELDS': ['pconsume', 'pconsumecounter:max', 'psupply', 'psupplycounter:max'], 'THRESHOLD': 80, 'RECONNECT_INTERVAL': 86400, 'DEBUG': False}`

kristian, 2025-01-13:
If I try to remove the empty array [] in the YAML and try to save in the Home Assistant UI, I get the error that:

Fehler beim Speichern der Add-on Konfiguration, Missing option 'AUTODISCOVER_CONFIG' in root in SMA Energy Meter

kellerza, 2025-01-13:
Can you try the new version?

kristian, 2025-01-13:
Could you do a quick bump, otherwise my Home Assistant doesn't provide me the option to update in the UI. thanks!

kellerza, 2025-01-13:
You should refresh the store

app-ch, 2025-01-13:
next error:

2025-01-13 14:56:36,804 INFO    options: Loading HASS OS configuration
2025-01-13 14:56:36,811 INFO    MQTT: Connecting to mqtt@core-mosquitto:1883
2025-01-13 14:56:36,832 INFO    MQTT: Connection successful
could not connect to mulicast group or bind to given interface
[14:56:37] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

kristian, 2025-01-13:
Tried restarting the HA instance, but didn't work.

kellerza, 2025-01-13:
Ensure that ipbind & multicast groups are ok,- similar to the defaults.

Can you paste your full config?

kristian, 2025-01-13:
```
MCASTGRP: 239.12.255.254
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt
MQTT_PASSWORD: xxx
IPBIND: ""
SMA_SERIALS:
  - "301xxx“
FIELDS:
  - pconsume
  - pconsumecounter:max
  - psupply
  - psupplycounter:max
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: false
AUTODISCOVER_CONFIG: []
```

Except the MQTT credentials and SMA_SERIALS I didn‘t change the settings.

app-ch, 2025-01-13:
great it works again! Thanks

cambronf, 2025-01-13:
Since upgrade to 2025,1,13b with config:

```
MCASTGRP: 239.12.255.254
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: fuser
MQTT_PASSWORD: xxxxx
IPBIND: 0.0.0.0
SMA_SERIALS:
  - "300694xxxx"
  - "190141yyyy"
  - "190141zzzz"
FIELDS:
  - pconsumecounter:max
  - psupplycounter:max
  - psupply:60
  - pconsume:60
  - psupply:10
  - pconsume:10
  - frequency
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: false
AUTODISCOVER_CONFIG: []
```

Logging is:
```
2025-01-13 16:29:44,317 INFO    options: Loading HASS OS configuration
2025-01-13 16:29:44,322 INFO    MQTT: Connecting to fcambron@core-mosquitto:1883
2025-01-13 16:29:44,329 INFO    MQTT: Connection successful
2025-01-13 16:29:44,832 INFO    Listening for multicast frames. Sensor discovery will be triggered by the first frame.
+ warnings as:
2025-01-13 16:30:01,801 WARNING Serial 4294967295 not in list of SMA_SERIALS
2025-01-13 16:30:17,639 WARNING Serial 27833254 not in list of SMA_SERIALS
2025-01-13 16:30:31,509 WARNING Serial 24818511 not in list of SMA_SERIALS
```

None of the entities are updated anymore (state remains Unknown)
also after HA restart,

Any idea?

cambronf, 2025-01-13:
Small extra remark in release notes:
**2024.1.13** - 2024-05-10
Added Address to bind to as an option (useful if you have multiple addresses)
2024.5.10 - 2024-05-10

2024.1.13 should be 2025.1.13

kellerza, 2025-01-13:
You need an empty sma_serials or the serial number you are interested in

kristian, 2025-01-13:
> great it works again! Thanks

Same here, works again! Thanks!

cambronf, 2025-01-13:
@kellerza Working again with empty list. All 3 meters are discovered automatic.
Why is it not working, when I specify the 3 serial numbers in the field: SMA_SERIALS ?

Anyway thanks for your work on this

kristian, 2025-01-13:
> @kellerza Working again with empty list. All 3 meters are discovered automatic. Why is it not working, when I specify the 3 serial numbers in the field: SMA_SERIALS ?
> 
> Anyway thanks for your work on this

I think there is still a bug in the SMA_SERIALS list, see #50. Before the recent changes the list simply didn‘t do anything (nothing was filterer), now, due to the bug, the filters in the list don‘t work yet.

dikra3572, 2025-01-14:
Hello,
since the update today, the status classes are no longer available for all sensors.
The error message is displayed during repairs and I am suggested to delete them.
If I delete them, however, all long-term statistics are also deleted from the database.
How should I deal with this?
Many thanks and regards
Dieter

kristian, 2025-01-14:
Whilst I don‘t know if deleting the status classes (don‘t know what you mean by that, sorry, I guess you mean sensors?) is a good idea, as far as I know, HA doesn‘t remove statistical data, if you delete an entity. Meaning if you delete an entity and re-create it with the same ID, the statistics should still be available. I don‘t know if HA runs cleanup jobs or whatever, as said, I am not suggesting to you to delete it, just wanted to share my experience with the statistical values.

dikra3572, 2025-01-14:
Thanks for the feedback.

Here is the full error message:

_The entity no longer has a state class

In the past, statistics were generated for "SMA Energy Meter qconsume" (sensor.sma_energy_meter_qconsume), but it no longer has a state class and therefore long-term statistics can no longer be created for it.

Statistics cannot be created until this entity has a supported state class.

If the state class was previously provided by an integration, this may be a bug. Please report the issue.

If you previously set the state class yourself, please correct it. The different state classes and their use can be found in the developer documentation.
If the state class has been permanently removed, you may want to delete the corresponding long-term statistics from your database.

Do you want to permanently delete the long-term statistics of sensor.sma_energy_meter_qconsume from your `database?_

kellerza, 2025-01-14:
I doubt it ever had a "state_class", but might have been triggered by the mqtt_entity update.

Try the latest dev version, where you now have to specify the serial and ha_prefix in `SMA_DEVICES`

dikra3572, 2025-01-14:
Thanks for the feedback but how do I install the latest dev version?

kellerza, 2025-01-14:
The repo contains three addons
1. SMA Energy Meter
2. SMA Energy Meter (developer version)
3. SMA Energy Meter (legacy version)

Install the 2nd one and make sure to copy your config and **stop the 1st BEFORE** starting the 2nd!

dikra3572, 2025-01-14:
That's too risky for me.
I'd rather wait until it's official.

dikra3572, 2025-01-14:
I overlooked that this topic already exists. So the error is probably not new.

[https://github.com/kellerza/hassio-sma-em/issues/47#issue-2680482795](url)

kristian, 2025-01-14:
@kellerza seems to be working fine for me in the dev version! Also log spamming reported in #50 seemed to have stopped with the new SMA_DEVICES configuration. For a compatible switch of one device having "SMA" as the prefix of a single device is correct, right? As soon as you release this to the normal branch, I will switch from the dev. to the normal version again.

kellerza, 2025-01-14:
The ha_prefix should be part of the entity name in home assistant

```
sensors.(ha_prefix)_psupply
```

Except maybe if you renamed these in the past (then it will stay the same due to the serial number used in the entity's ID

kristian, 2025-01-14:
Make sense and works for me! I used the prefix `SMA Energy Meter" now, which converts to the same sensor name `sensor.sma_energy_meter_psupply", but as you said, it did map to the same entity ID as before anyways, so all is good! 👍

kristian, 2025-01-14:
@kellerza  Do you plan to still move the changes of the DEV version to the release version to complete this and #50? Thanks.

kellerza, 2025-01-14:
Let’s wait for another day or two then I’ll move it there

New config options, so don't want to do it too much

kristian, 2025-01-14:
Sure, I will stay on the DEV version until then and report if there is any issues. Thanks for your support @kellerza, highly appreciated 👍