kellerza/hassio-sma-em #52: After updating to 2025.1.14 HA claims that state classes are missing for some properties

igotsoul, 2025-01-15:
Please see the screenshot, yaml config attached below. It´s only for those 3 fields, the others doesn't seem to be affected. Any ideas?

[image]

```
FIELDS:
  - pconsumecounter:max
  - psupplycounter:max
  - pconsume:1
  - psupply:1
  - u1:1
  - u2:1
  - u3:1
  - i1:1
  - i2:1
  - i3:1
  - p1consume:1
  - p2consume:1
  - p3consume:1
  - p2supply:1
  - p3supply:1
MCASTGRP: 239.12.255.254
IPBIND: 0.0.0.0
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: 0
AUTODISCOVER_CONFIG: []
```

kellerza, 2025-01-15:
Can you try the developer version?
The config is slightly different

igotsoul, 2025-01-15:
Sure, can try it. There's by the way another thing that caught my eye just recently. After the add-on is running for some time it reports cast errors in the log:

```
2025-01-15 11:08:19,853 INFO    options: Loading HASS OS configuration
2025-01-15 11:08:19,854 DEBUG   options: Loading config: {'MQTT_HOST': 'core-mosquitto', 'MQTT_PORT': 1883, 'MQTT_USERNAME': 'xxxx', 'MQTT_PASSWORD': 'xxxxx', 'SMA_SERIALS': ['xxxxxxx'], 'FIELDS': ['pconsumecounter:max', 'psupplycounter:max', 'pconsume:1', 'psupply:1', 'u1:1', 'u2:1', 'u3:1', 'i1:1', 'i2:1', 'i3:1', 'p1consume:1', 'p2consume:1', 'p3consume:1', 'p2supply:1', 'p3supply:1'], 'MCASTGRP': '239.12.255.254', 'IPBIND': '0.0.0.0', 'THRESHOLD': 80, 'RECONNECT_INTERVAL': 86400}
2025-01-15 11:08:19,873 INFO    MQTT: Connecting to mqtt_user@core-mosquitto:1883
2025-01-15 11:08:19,890 INFO    MQTT: Connection successful
2025-01-15 11:08:20,392 INFO    Listening for multicast frames. Sensor discovery will be triggered by the first frame.
2025-01-15 11:08:21,388 INFO    Multicast frame received for SMA serial 3004904911, discovered sensors:
2025-01-15 11:08:21,389 INFO     - pconsumecounter_max every 60s
2025-01-15 11:08:21,389 INFO     - psupplycounter_max every 60s
2025-01-15 11:08:21,389 INFO     - pconsume_1 every 1s
2025-01-15 11:08:21,390 INFO     - psupply_1 every 1s
2025-01-15 11:08:21,390 INFO     - u1_1 every 1s
2025-01-15 11:08:21,390 INFO     - u2_1 every 1s
2025-01-15 11:08:21,391 INFO     - u3_1 every 1s
2025-01-15 11:08:21,391 INFO     - i1_1 every 1s
2025-01-15 11:08:21,391 INFO     - i2_1 every 1s
2025-01-15 11:08:21,391 INFO     - i3_1 every 1s
2025-01-15 11:08:21,392 INFO     - p1consume_1 every 1s
2025-01-15 11:08:21,392 INFO     - p2consume_1 every 1s
2025-01-15 11:08:21,392 INFO     - p3consume_1 every 1s
2025-01-15 11:08:21,392 INFO     - p2supply_1 every 1s
2025-01-15 11:08:21,393 INFO     - p3supply_1 every 1s
2025-01-15 11:08:22,244 WARNING Serial 4294967295 not in list of SMA_SERIALS
unknown datatype: measurement 29697 datatype unknown raw_type 207
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 29697 datatype unknown raw_type 207
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 29697 datatype unknown raw_type 207
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 29697 datatype unknown raw_type 207
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
```

aheinzig, 2025-01-15:
The same here. After everything has booted up properly, two more sensors are detected that I don't have. One serial is the same of the post from igotsoul.
[image]
[image]

aheinzig, 2025-01-15:
First of all, a big thank you to kellerza for this great work! I have a similar problem in the DEV version after the update to 2025.1.14
After everything has booted up properly, two more sensors are detected that I don't have. One of the serials is the same from the post of igotsoul. Then, a few minutes later, the same error reports in the log.

[image]
[image]

kellerza, 2025-01-15:
state_class is a duplicate of #47

closed 2025-01-15 by kellerza

aheinzig, 2025-01-15:
You are so great. After the update to 2025.1.14b the errors are gone. A million thanks!

[image]

dikra3572, 2025-01-16:
Hello,
after updating to version 2025.1.15 the problem unfortunately still exists.

kellerza, 2025-01-16:
> Hello, after updating to version 2025.1.15 the problem unfortunately still exists.

There are multiple problems in this issue, so best to start a new issue & post some logs