kellerza/hassio-sma-em #54: "Unknown datatype" messages in LOG since update to 2025.1.14

MichaelsSynology, 2025-01-15:
Hi,
the above update fixed the issue of the messages coming from other SMA devices as the serial numbr filtering now works well. Thank you for that!

But, with this update I now have those messages left spaming my log:

**unknown datatype: measurement 29697 datatype unknown raw_type 201
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0
unknown datatype: measurement 0 datatype unknown raw_type 0**

Those I did not had with the version before.
Any idea what is wrong?

Michael

kellerza, 2025-01-15:
Should be fixed in #52

closed 2025-01-15 by kellerza

MichaelsSynology, 2025-01-16:
Fixed with version 2025.1.15.
Thank you a lot for all your efforts.

One small hint:
With version 2025.1.15 the methode to enter the serial number of your SMA Home Manager has changed. So, there is a need to enter again this serial number in configuration tab.

kellerza, 2025-01-16:
Thanks

The change was needed to filter other SMA devices and to assign a consistent and unique ha_prefix when you have multiple SMA devices