kellerza/hassio-sma-em #55: Serial Number 007 since last update

asmcc, 2025-01-22:
Since last update (I assume since 2-3 days) the serial number of my EM was 007. I do not know, why. Thereafter this integration does not work, all values were grey. The main issue is, that the user WILL NOT BE INFORMED about this!
I sensed it only today as I saw, that energy values in HA are wrong and then, that all EM entities are grey.

My awaiting for such casesis , that the integration does not start. But that was not the case here! The integration starts and show a normal running state!
Only the search in the protocol show me this error message:
`2025-01-22 19:57:31,723 WARNING Unknown SMA serial number XXXXXXXXX. If you want to use this device,please add SERIAL_NR and HA_PREFIX to SMA_DEVICES
2025-01-22 19:57:50,825 WARNING Unknown SMA serial number YYYYYYYYY. If you want to use this device,please add SERIAL_NR and HA_PREFIX to SMA_DEVICES`
First one is my EM. That is ok. The second one? I do not know the number.
In the properties I found, that the serial nuber is "007". But I did not changed this number to "007"

I corrected the serial number from 007 to my own and after it all works fine. But I want to report this issue here.

Please make your intergation more backward compatible, if you make such radical changes. Or if you initalise / load here a default value (like this 007), then mark the integration so, that it can not start and MUST BE NEW INITIALISED. Or A RED BOLD WARNING to the user: TO CHECK AND INPUT THE SERIAL NUMBER AFTER THIS UPDATE

Thanks!

kellerza, 2025-01-23:
SMA_DEVICES with 007 would only appear once you updated & **reset the config to defaults**. 

Please pay attention to the changelog when you do this.

closed 2025-01-23 by kellerza

asmcc, 2025-01-23:
I did not reseted it to defaults!
All other parameters were not changed, only this SMA_DEVICES, since you defined a new variable now and do not translate the serial number from the old variable to the new one. We are here in the development version and I understand it. But I want to preserve you from the shitshtorm an issues here, if you transfer it into stable version.
And you ignorred also my second comment. Usualy a not configured intergation is marked as such one under Home Assistant. That is the behaviour, which I know from all other integrations / devices etc. If the integration is marked as not configured, then it does not start and you can see it as user. That was definitivelly not the case here.
Therefore you can ignore my warnings here and close this issue here. Then you won't have to be surprised later. You are warned

kellerza, 2025-01-24:
If you want config to be translated, that is outside the addon's control and only something the HA supervisor can do. (same with getting that 007 default)