kellerza/hassio-sma-em #57: Devices are not recognized after update to 2025.1.15.

Sab3rRid3r, 2025-01-26:
Hi, I have been using the addon since 2021 with two Energy Meters and currently in version 2025.1.14.
So far I have not specifically mentioned the devices in the configuration.

My Config:
MQTT_HOST: 192.168.xxx.yyy
MQTT_PORT: 1883
MQTT_USERNAME: xxxxxxxxx
MQTT_PASSWORD: yyyyyyyyy
SMA_SERIALS: []
FIELDS:
  - pconsume
  - pconsumecounter:max
  - psupply
  - psupplycounter:max
MCASTGRP: 239.12.xxx.yyy
IPBIND: 0.0.0.0
THRESHOLD: 80
RECONNECT_INTERVAL: 86400
DEBUG: 0
AUTODISCOVER_CONFIG: []

The data is currently published via MQTT as follows eg.
SMA-EM/status/1901417xxx/pconsume
SMA-EM/status/1901403yyy/pconsume

After updating to version 2025.1.15, I receive the following error message.
Unknown SMA serial number 1901417xxx. If you want to use this device,please add SERIAL_NR and HA_PREFIX to SMA_DEVICES
Unknown SMA serial number 1901403yyy. If you want to use this device,please add SERIAL_NR and HA_PREFIX to SMA_DEVICES

What must the configuration look like so that it corresponds 1:1 to the existing configuration and no new devices or entities are created?
 SMA_DEVICES:
    - SERIAL_NR: 1901417xxx
      PREFIX: sma ???
    - SERIAL_NR: 1901403yyy
      PREFIX: sma2 ???

Thank you very much

kellerza, 2025-01-26:
The prefix is important during initial discovery. Most likely it wont make a difference for you. 

Depending on where you come from the unique id didn't change, so the entities are the same

Sab3rRid3r, 2025-01-27:
Hi Johann, the configuration works and no new ids have been created. Thank you very much!

SMA_DEVICES:
  - SERIAL_NR: "1901417xxx"
    HA_PREFIX: waermepumpe
  - SERIAL_NR: "1901403yyy"
    HA_PREFIX: netz