kellerza/hassio-sma-em #73: new version of SMA EM crashes within a few seconds

steerage250, 2025-07-15:
The add-on stopped working at 3:15am 17 July (GMT+8) - seemingly after an update??  The watchdog was off (not sure if it had always been off - I usually have them on).  The watchdog was seemingly not restarting it (after I turned the watchdog on).

The logfile presents like this:
09:41:03 ERROR   MQTT: Home Assistant not online. Topic homeassistant/status is empty
[09:41:03] WARNING: [33mHalt add-on with exit code 1[0m
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Sometimes the watchdog seems to be able to restart it, and sometimes not.  Typically runs for about 10 seconds and updates data, then stops as above.

MQTT Explorer shows other MQTT devices still updating, but  SMA-EM status is offline

kellerza, 2025-07-15:
Please have a look at the other issues before posting a new one

steerage250, 2025-07-15:
My apologies

closed 2025-07-15 by steerage250