kellerza/hassio-sma-em #77: Latest update ignores MQTT config and only looks for the core-mosquitto host

matqua, 2025-07-23:
Could there be an issue with the latest updates (July) that causes the addon to ignore or not process the user's MQTT config and only use core-mosquitto as the MQTT_HOST (possibly also an issue with the other MQTT_* parameters?)

I did a quick compare of the options.py from 6 months ago and now and saw that a number of the mqtt_* options are no longer present. I am not competent enough in python/docker to understand if the way these are processed now are somewhere else in the code.

- I'm running the EMQX broker add-on rather than the default core-mosquitto one. 
- I updated my sma-em add-on config with the emqx container name a0d7b954-emqx for the MQTT_HOST and other MQTT_* parameters as required.
- This add-on worked fine for a while. 
- Last couple of weeks, noticed no solar data on my dashboard, finally look into it today and check the logs and it has errors connecting to the MQTT broker.

```
[05:56:50] INFO: [32mStarting the SMA-EM add-on[0m
s6-rc: info: service legacy-services successfully started
[05:56:51] [32mINFO   [0m Loading config: /data/options.json
[05:56:51] [32mINFO   [0m MQTT: Connecting to addons@core-mosquitto:1883
[05:56:51] [32mINFO   [0m Listening for frames on interface 0.0.0.0, for multicast group 239.12.255.254
[05:56:51] [32mINFO   [0m Sensor discovery will be triggered by the first frame containing these serial numbers: 26784661, 007, 3011950729, 4294967295
[05:56:52] [32mINFO   [0m Multicast frame received for SMA serial 3011950729, discovered sensors:
[05:56:52] [32mINFO   [0m  - pconsume every 60s
[05:56:52] [32mINFO   [0m  - pconsumecounter_max every 60s
[05:56:52] [32mINFO   [0m  - psupply every 60s
[05:56:52] [32mINFO   [0m  - psupplycounter_max every 60s
[05:56:52] [32mINFO   [0m  - frequency every 60s
[05:56:56] [31mERROR  [0m MQTT: Connection timeout (5s)
[05:56:56] [31mERROR  [0m Task exception was never retrieved
```

I then see the connecting line:
```
MQTT: Connecting to addons@core-mosquitto:1883
```
which is wrong. My config is 

```yaml
SMA_DEVICES:
  - SERIAL_NR: "30119ABCDE"
    HA_PREFIX: SMA
  - SERIAL_NR: "42949ABCDE"
    HA_PREFIX: Z_7SMA429
FIELDS:
  - pconsume
  - pconsumecounter:max
  - psupply
  - psupplycounter:max
  - frequency
MQTT_HOST: a0d7b954-emqx
MQTT_PORT: 1883
MQTT_USERNAME: sma
MQTT_PASSWORD: <SUPERDOOPERSECRETPASSWORD>
MCASTGRP: 239.12.255.254
IPBIND: 0.0.0.0
THRESHOLD: 80
RECONNECT_INTERVAL: 3600
DEBUG: false
SMA_SERIALS: []
AUTODISCOVER_CONFIG: []

```

Rolled back to the previous version and it is all working again.

```
You have 189fd64 installed. Press update to update to version 1edb86e
```

kellerza, 2025-07-27:
These setting comes directly from the Home Assistant supervisor
<https://github.com/kellerza/mqtt_entity/blob/main/src/mqtt_entity/options.py#L114>

Do you run mosquitto as well?
What is configured in Home Assistant?

matqua, 2025-07-27:
I used to run mosquitto. The addon is still there but stopped/disabled.

Home assistant is configured and is connecting to the running EMQX addon:

[image]

matqua, 2025-07-27:
Thank you very much for your response!

I checked my supervisor api mqtt endpoint and it indeed was showing the core-mosquitto one listed. I fully deleted the mosquitto addon (the supervisor api would not let me execute the delete method/call) and the latest version of your addon now works.

The logs show it queries the supervisor for the broker, fails, then uses the mqtt_* defined config for the emqx addon.

You again seem to be right re the emqx addon not registering itself as the broker in the supervisor. I found this https://github.com/hassio-addons/addon-emqx/issues/132 which seems like it may never register itself.

Many thanks again for your time.

closed 2025-07-27 by matqua